sarxos/webcam-capture #882: Support Java 17

ScottPierce, 2022-11-02:
Seems like this doesn't work with newer versions of Java, i.e. 17. It's likely an issue with the underlying BridJ library. Every time I run this I get the error:

```
# BridJ: LoadLibrary error when loading C:\Users\spier\AppData\Local\Temp\BridJExtractedLibraries14486809914425661988\OpenIMAJGrabber.dll : The specified module could not be found.
```

alexmao86, 2022-11-21:
@ScottPierce , thank you for your interests, this lib was developed under JDK8, JDK 11 and before are compatible according to open source community experience. Since from JDK 14, there are very good new features on native access, there is no plan to support higher JDK. You can follow the framework, just a simple Driver can done your work, I think
thanks

alexmao86, 2022-11-21:
sorry for inconvenience.

closed 2022-11-21 by alexmao86

ScottPierce, 2022-11-22:
> just a simple Driver can done your work

@alexmao86 What do you mean by this? This can't possibly be true.