sarxos/webcam-capture #883: How to Get Button Press Event OF UVC Device

mohammadwasim91, 2022-11-15:
Hi
I am using this library to get a preview of the UVC device.
I want to capture the image, do some processing and save it on the disc, I have added a button on the screen, and on clicking that, I am taking the snapshot and saving it onto the file.
The same operations should also be performed when I press the SCAN button present on the UVC device but I am not aware the how can I listen to the button press event of the camera.

eduramiba, 2022-11-15:
Hi @mohammadwasim91 

You will need to create a custom library for the UVC extension of this device. Do you have documentation from the vendor? Which device is it?

Which OS do you intend to use?

alexmao86, 2022-11-21:
Hi,
webcam-capture is trying to capture webcam to java swing BufferredImage, even without audio. so regarding buttons on device, it is out of webcam-capture scope. Just like @eduramiba comments, it is highly depends on your device, there shall be some other open source with JNI JNA support to work with hardware outside only if there are interface on your device.

thanks

closed 2022-11-21 by alexmao86