sarxos/webcam-capture #896: How to query for webcam capabilities?

polentino, 2023-01-07:
Hi all,
when I run a simple application on a RPI Zero W, i see the following output to appear:
```
Does NOT have capture capability
Does NOT have output capability
Does NOT have overlay capability
Does NOT have VBI capture capability
Does NOT have VBI output capability
Does NOT have SLICED VBI capture capability
Does NOT have SLICED VBI output capability
Does NOT have RDS capability
Does NOT have tuner capability
Does NOT have audio capability
Does NOT have radio capability
Does NOT have read/write capability
Does NOT have asyncIO capability
Has streaming capability
```
which made me wonder: is there a specific object I can query, to get which capabilities are actually available on the Webcam I'm trying to open? It would be nice to know in advance, for example, whether the webcam im using is capable of capturing still images, or streaming them. Thanks!

shinobisoft, 2023-02-04:
I don't know about USB cameras, but for IP camera's, you send requests to the server via http get/post to get information about it's capabilities.

giftest, 2023-04-24:
Hi Sarxos,
what is the database of this project "Group-Chat-App-With-Encrypted-Messaging-and-Video-Streaming-main"
Please can you share the database with me because I want to see where the data are stored?
Thank you in advance, Man.