sarxos/webcam-capture #905: Strange Trouble (not an issue but i can't work with your wonderful API)

davideserra, 2023-07-23:
Hi, 
First of all my compliment for your job: really great!!

I'm making a sub rover with raspberry PI P4 64 bit and I really need to implement a cam system to drive it and watch where it is. Unfortunately I find e problem using slog4j because every time I try (in my software) Webcam webcam = Webcam.getDefault() I get a very strange exception don't have anything to do with your software but prevent me to go ahed:

Exception in thread "AWT-EventQueue-0" java.lang.NoClassDefFoundError: ch/qos/logback/core/joran/spi/JoranException
	at org.slf4j.LoggerFactory.getSingleton(LoggerFactory.java:189)
	at org.slf4j.LoggerFactory.bind(LoggerFactory.java:112)
	at org.slf4j.LoggerFactory.performInitialization(LoggerFactory.java:105)
	at org.slf4j.LoggerFactory.getILoggerFactory(LoggerFactory.java:235)
	at org.slf4j.LoggerFactory.getLogger(LoggerFactory.java:208)
	at org.slf4j.LoggerFactory.getLogger(LoggerFactory.java:221)
	at com.github.sarxos.webcam.Webcam.<clinit>(Webcam.java:101) 

Have you notice of something like that? I imported on my netbeans project all available library on the net so far but I didn't solve the issue. 

Thanks so much in advance and best regards,
Davide

C-Dykstra, 2023-09-21:
I get a very similar error:
Exception in thread "main" java.lang.NoClassDefFoundError: org/slf4j/LoggerFactory
	at com.github.sarxos.webcam.Webcam.<clinit>(Webcam.java:101)
	at Tester.main(Tester.java:17)
Caused by: java.lang.ClassNotFoundException: org.slf4j.LoggerFactory
	at java.base/jdk.internal.loader.BuiltinClassLoader.loadClass(BuiltinClassLoader.java:641)
	at java.base/jdk.internal.loader.ClassLoaders$AppClassLoader.loadClass(ClassLoaders.java:188)
	at java.base/java.lang.ClassLoader.loadClass(ClassLoader.java:525)
	... 2 more

alexmao86, 2023-12-05:
You project is not mananged by maven, so the slf4j and dedefault logback logger IMPL are missing, please copy coresponding jars to your classpath. ti should work out.

closed 2023-12-05 by alexmao86