sarxos/webcam-capture #907: Kotlin Multiplatform Desktop MacOS

codaman, 2023-08-31:
I am trying to implement Kotlin Compose Multiplatform for MacOS. There is no webcam sample related Kotlin Multiplatform. 

I tried issue #757 for macOS. I used this solution in this issue: [solution_link](https://github.com/sarxos/webcam-capture/issues/757#issuecomment-849632328)

My Gradle file is like this:

repositories part:

```
maven {
        url = uri("http://oss.sonatype.org/content/repositories/snapshots")
        isAllowInsecureProtocol = true
    }
    maven {
        url = uri("http://maven.openimaj.org")
        isAllowInsecureProtocol = true
    }

    mavenCentral {
        url = uri("https://maven.ecs.soton.ac.uk/content/groups/maven.openimaj.org")
        isAllowInsecureProtocol = true
    }
```

dependencies part:

```
implementation ("com.github.sarxos:webcam-capture:0.3.13-SNAPSHOT")
implementation("com.github.sarxos:webcam-capture-driver-openimaj:0.3.13-SNAPSHOT") {
        exclude(group = "org.openimaj", module = "core-video-capture")
}
 implementation ("com.nativelibs4java:bridj:0.7-20140918-3")
```

but when compiling I got 
warning: default scripting plugin is disabled: The provided plugin org.jetbrains.kotlin.scripting.compiler.plugin.ScriptingCompilerConfigurationComponentRegistrar is not compatible with this version of compiler
error: unable to evaluate script, no scripting plugin loaded

if I tried 0.3.12 released version, there is no compile error but webcam not working and crashed in runtime. in 0.3.13-SNAPSHOT version I got this compile error

codaman, 2023-09-01:
I upgraded jvm. Fix that issue. But now I got.

SLF4J: Failed to load class "org.slf4j.impl.StaticLoggerBinder".
SLF4J: Defaulting to no-operation (NOP) logger implementation
SLF4J: See http://www.slf4j.org/codes.html#StaticLoggerBinder for further details.
Exception in thread "AWT-EventQueue-0" com.github.sarxos.webcam.WebcamException: java.util.concurrent.ExecutionException: com.github.sarxos.webcam.WebcamException: Cannot execute task
	at com.github.sarxos.webcam.WebcamDiscoveryService.getWebcams(WebcamDiscoveryService.java:124)
	at com.github.sarxos.webcam.Webcam.getWebcams(Webcam.java:893)
	at com.github.sarxos.webcam.Webcam.getWebcams(Webcam.java:866)
	at com.github.sarxos.webcam.Webcam.getWebcams(Webcam.java:845)
	at MainKt.main$lambda$8(Main.kt:312)
	at java.desktop/java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:318)
	at java.desktop/java.awt.EventQueue.dispatchEventImpl(EventQueue.java:773)
	at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:720)
	at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:714)
	at java.base/java.security.AccessController.doPrivileged(AccessController.java:399)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:86)
	at java.desktop/java.awt.EventQueue.dispatchEvent(EventQueue.java:742)
	at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:203)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:124)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:113)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:109)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.desktop/java.awt.EventDispatchThread.run(EventDispatchThread.java:90)
Caused by: java.util.concurrent.ExecutionException: com.github.sarxos.webcam.WebcamException: Cannot execute task
	at java.base/java.util.concurrent.FutureTask.report(FutureTask.java:122)
	at java.base/java.util.concurrent.FutureTask.get(FutureTask.java:191)
	at com.github.sarxos.webcam.WebcamDiscoveryService.getWebcams(WebcamDiscoveryService.java:116)
	... 17 more
Caused by: com.github.sarxos.webcam.WebcamException: Cannot execute task
	at com.github.sarxos.webcam.WebcamProcessor$AtomicProcessor.process(WebcamProcessor.java:72)
	at com.github.sarxos.webcam.WebcamProcessor.process(WebcamProcessor.java:140)
	at com.github.sarxos.webcam.WebcamTask.process(WebcamTask.java:46)
	at com.github.sarxos.webcam.ds.buildin.WebcamDefaultDriver$GetDevicesTask.getDevices(WebcamDefaultDriver.java:79)
	at com.github.sarxos.webcam.ds.buildin.WebcamDefaultDriver.getDevices(WebcamDefaultDriver.java:124)
	at com.github.sarxos.webcam.WebcamDiscoveryService$WebcamsDiscovery.call(WebcamDiscoveryService.java:36)
	at com.github.sarxos.webcam.WebcamDiscoveryService$WebcamsDiscovery.call(WebcamDiscoveryService.java:26)
	at java.base/java.util.concurrent.FutureTask.run(FutureTask.java:264)
	at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1136)
	at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:635)
	at java.base/java.lang.Thread.run(Thread.java:833)
Caused by: java.lang.UnsatisfiedLinkError: 'org.bridj.Pointer com.github.sarxos.webcam.ds.buildin.natives.OpenIMAJGrabber.getVideoDevices()'
	at com.github.sarxos.webcam.ds.buildin.natives.OpenIMAJGrabber.getVideoDevices(Native Method)
	at com.github.sarxos.webcam.ds.buildin.WebcamDefaultDriver$GetDevicesTask.handle(WebcamDefaultDriver.java:93)
	at com.github.sarxos.webcam.WebcamProcessor$AtomicProcessor.run(WebcamProcessor.java:81)
	... 3 more

akexorcist, 2023-10-27:
It works fine with my Compose Desktop project on macOS. You can see my code from https://github.com/akexorcist/backdrop

codaman, 2024-01-25:
Thank you so much @akexorcist. It is working. You saved my project