sarxos/webcam-capture #911: dont render well

christian-llauderes96, 2023-11-03:
I get some example here(WebcamPanelExample). In  netbeans it works fine but when I build it as executable file and open it works except for the camera. It seems very slow first it was "device initialization" after a plenty seconds (33s) it shows but black video and repaint value is 0.2 and fps 0.0 and doesn't change. I don't know what is the problem.

eduramiba, 2023-11-16:
I suggest you try this driver: https://github.com/eduramiba/webcam-capture-driver-native

christian-llauderes96, 2023-11-16:
> I suggest you try this driver: https://github.com/eduramiba/webcam-capture-driver-native

will this work in java ant?? My project was not in maven

eduramiba, 2023-11-17:
Sure it should work if you download the jars from maven central: https://repo1.maven.org/maven2/io/github/eduramiba/webcam-capture-driver-native/1.0.0/

Then, if using windows, remember to copy the correct DLLs from https://github.com/eduramiba/webcam-capture-driver-native/tree/master/natives to where your program will run

christian-llauderes96, 2023-12-05:
> I get some example here(WebcamPanelExample). In netbeans it works fine but when I build it as executable jar file and open it works except for the camera. It seems very slow first it was "device initialization" after a plenty seconds (33s) it shows but black video and repaint value is 0.2 and fps 0.0 and doesn't change. I don't know what is the problem.
> 
> In desktop works fine. The problem arises on laptop

In logs
2023-12-05 09:57:24,183 [AWT-EventQueue-0] INFO  com.github.sarxos.webcam.Webcam - WebcamDefaultDriver capture driver will be used
2023-12-05 09:57:24,353 [atomic-processor-1] INFO  com.github.sarxos.webcam.ds.cgt.WebcamOpenTask - Opening webcam USB2.0 HD UVC WebCam 0
2023-12-05 09:57:57,730 [atomic-processor-1] ERROR com.github.sarxos.webcam.ds.buildin.WebcamDefaultDevice - Timeout when requesting image!
2023-12-05 09:58:03,225 [frames-refresher-[0]] ERROR com.github.sarxos.webcam.ds.buildin.WebcamDefaultDevice - Timeout when requesting image!
2023-12-05 09:58:08,814 [atomic-processor-1] ERROR com.github.sarxos.webcam.ds.buildin.WebcamDefaultDevice - Timeout when requesting image!
2023-12-05 09:58:14,381 [frames-refresher-[0]] ERROR com.github.sarxos.webcam.ds.buildin.WebcamDefaultDevice - Timeout when requesting image!