sarxos/webcam-capture #913: Opening OBS Virtual Camera crashes app with "ERROR: SampleCB() - buffer sizes do not match"

i-make-robots, 2023-11-13:
Install OBS, which creates OBS Virtual Camera (https://obsproject.com/)
Use Webcam to detect camera.
Attempt to open camera stream of OBS Virtual Camera [any number] at resolution.
```
07:38:47.145 [webcam-discovery-service] DEBUG c.g.s.w.d.b.WebcamDefaultDriver - Searching devices
07:38:47.146 [webcam-discovery-service] DEBUG c.g.s.w.d.b.WebcamDefaultDriver - Found device HD Pro Webcam C920 0
07:38:47.146 [webcam-discovery-service] DEBUG c.g.s.w.d.b.WebcamDefaultDriver - Found device OBS Virtual Camera 1
07:38:49.405 [AWT-EventQueue-0] DEBUG com.github.sarxos.webcam.Webcam - Setting new resolution 640x480
07:38:49.406 [AWT-EventQueue-0] DEBUG c.github.sarxos.webcam.WebcamLock - Lock Webcam OBS Virtual Camera 1
07:38:49.408 [atomic-processor-1] INFO  c.g.s.webcam.ds.cgt.WebcamOpenTask - Opening webcam OBS Virtual Camera 1
07:38:49.408 [atomic-processor-1] DEBUG c.g.s.w.d.b.WebcamDefaultDevice - Opening webcam device OBS Virtual Camera 1
07:38:49.408 [atomic-processor-1] DEBUG c.g.s.w.d.b.WebcamDefaultDevice - Webcam device 1 starting session, size java.awt.Dimension[width=640,height=480]
07:38:49.503 [atomic-processor-1] DEBUG c.g.s.w.d.b.WebcamDefaultDevice - Webcam device session started
07:38:49.504 [atomic-processor-1] DEBUG c.g.s.w.d.b.WebcamDefaultDevice - Clear memory buffer
07:38:50.150 [webcam-discovery-service] DEBUG c.g.s.w.d.b.WebcamDefaultDriver - Searching devices
ERROR: SampleCB() - buffer sizes do not match
ERROR: SampleCB() - buffer sizes do not match
...
ERROR: SampleCB() - buffer sizes do not match
ERROR: SDisconnected from the target VM, address: '127.0.0.1:53445', transport: 'socket'
ampleCB() - buffer sizes do not match
....
```
version 0.3.12
java 21
intelliJ IDEA

i-make-robots, 2023-11-16:
Also tested with https://raw.githubusercontent.com/sarxos/webcam-capture/master/webcam-capture/src/example/java/WebcamViewerExample.java
same results.  First camera works fine, switch to OBS and boom.
Also tested while OBS is running and virtual camera streaming is off.  same result.
Also tested while OBS is running and virtual camera streaming is on.  same result.

eduramiba, 2023-11-16:
I suggest you try this driver: https://github.com/eduramiba/webcam-capture-driver-native

i-make-robots, 2023-11-16:
@eduramiba sounds good, but no linux support?  Any other option?

eduramiba, 2023-11-17:
Sorry no linux support but I guess the best option will be adding V4L2 to my driver.
It's hit and miss though, I found it unstable in older linux versions such as ubuntu 18, but did not try it much

eduramiba, 2025-11-25:
@i-make-robots @TureBentzin Can you test with latest version (1.2.2)? - io.github.eduramiba:webcam-capture-driver-native:1.2.2

It should now work with linux