sarxos/webcam-capture #930: Webcam Capture API not working in Mac with Arm architecture

namrathalokesh19, 2024-06-04:
On using default driver of Webcam Capture API, not able to access the webcam of Macbook pro with M1 architecture. It is failing to initialise the BridJ with below error. 

java.lang.RuntimeException: Failed to initialize BridJ (java.lang.UnsatisfiedLinkError: Can't load library: /var/folders/xf/t03s0z0n0f39m0rqld70wf380000gn/T/BridJExtractedLibraries16145624487360412057/libbridj.dylib)
	at org.bridj.Platform.initLibrary(Platform.java:444)
	at org.bridj.Platform.<clinit>(Platform.java:228)
	at com.github.sarxos.webcam.ds.buildin.natives.OpenIMAJGrabber.<clinit>(OpenIMAJGrabber.java:59)
	at com.github.sarxos.webcam.ds.buildin.WebcamDefaultDriver$WebcamNewGrabberTask.handle(WebcamDefaultDriver.java:55)
	at com.github.sarxos.webcam.WebcamProcessor$AtomicProcessor.run(WebcamProcessor.java:81)
	at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1144)
	at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:642)
	at java.base/java.lang.Thread.run(Thread.java:1589)
Caused by: java.lang.UnsatisfiedLinkError: Can't load library: /var/folders/xf/t03s0z0n0f39m0rqld70wf380000gn/T/BridJExtractedLibraries16145624487360412057/libbridj.dylib
	at java.base/java.lang.ClassLoader.loadLibrary(ClassLoader.java:2408)
	at java.base/java.lang.Runtime.load0(Runtime.java:785)
	at java.base/java.lang.System.load(System.java:2009)
	at org.bridj.Platform.initLibrary(Platform.java:420)

eduramiba, 2024-06-04:
> at org.bridj.Platform.(Platform.java:228)
> com.github.sarxos.webcam.ds.buildin.natives.OpenIMAJGrabber.(OpenIMAJGrabber.java:59)

I would recommend you try
https://github.com/eduramiba/webcam-capture-driver-native

SWSpratlin, 2024-09-02:
Confirming the NativeDriver works on Mac M1