sarxos/webcam-capture #939: Issues running on Raspberry Pi 5

SWSpratlin, 2024-09-02:
Using the V4L4J Driver, I consistently run into a "NoSuchFieldException: usr_paths"

I think it has something to do with loading the "loadLibraryFromJar" method, but I'm not experienced enough to really troubleshoot that. Seems like the "usr_paths" isn't getting created properly, potentially because of permissions, but I'm not totally sure. 

```
SLF4J: No SLF4J providers were found.
SLF4J: Defaulting to no-operation (NOP) logger implementation
SLF4J: See https://www.slf4j.org/codes.html#noProviders for further details.
java.io.IOException: java.lang.NoSuchFieldException: usr_paths
	at com.github.sarxos.v4l4j.NativeUtils.loadLibraryFromJar(NativeUtils.java:62)
	at com.github.sarxos.v4l4j.V4L4J.init(V4L4J.java:50)
	at com.github.sarxos.webcam.ds.v4l4j.V4l4jDriver.<clinit>(V4l4jDriver.java:33)
	at sarxtest.CamHandler.initDriver(CamHandler.java:34)
	at sarxtest.CamHandler.<init>(CamHandler.java:20)
	at sarxtest.HandlerTests.settings(HandlerTests.java:25)
	at processing.core.PApplet.handleSettings(PApplet.java:853)
	at processing.core.PApplet.runSketch(PApplet.java:10113)
	at sarxtest.HandlerTests.main(HandlerTests.java:87)
Caused by: java.lang.NoSuchFieldException: usr_paths
	at java.base/java.lang.Class.getDeclaredField(Class.java:2792)
	at com.github.sarxos.v4l4j.NativeUtils.addLibraryPath(NativeUtils.java:109)
	at com.github.sarxos.v4l4j.NativeUtils.loadLibraryFromJar(NativeUtils.java:60)
	... 8 more
com.github.sarxos.webcam.WebcamException: java.util.concurrent.ExecutionException: java.lang.UnsatisfiedLinkError: 'long au.edu.jcu.v4l4j.VideoDevice.doInit(java.lang.String)'
	at com.github.sarxos.webcam.WebcamDiscoveryService.getWebcams(WebcamDiscoveryService.java:124)
	at com.github.sarxos.webcam.Webcam.getWebcams(Webcam.java:893)
	at com.github.sarxos.webcam.Webcam.getWebcams(Webcam.java:866)
	at com.github.sarxos.webcam.Webcam.getWebcams(Webcam.java:845)
	at sarxtest.CamHandler.initCam(CamHandler.java:55)
	at sarxtest.HandlerTests.settings(HandlerTests.java:26)
	at processing.core.PApplet.handleSettings(PApplet.java:853)
	at processing.core.PApplet.runSketch(PApplet.java:10113)
	at sarxtest.HandlerTests.main(HandlerTests.java:87)
Caused by: java.util.concurrent.ExecutionException: java.lang.UnsatisfiedLinkError: 'long au.edu.jcu.v4l4j.VideoDevice.doInit(java.lang.String)'
	at java.base/java.util.concurrent.FutureTask.report(FutureTask.java:122)
	at java.base/java.util.concurrent.FutureTask.get(FutureTask.java:191)
	at com.github.sarxos.webcam.WebcamDiscoveryService.getWebcams(WebcamDiscoveryService.java:116)
	... 8 more
Caused by: java.lang.UnsatisfiedLinkError: 'long au.edu.jcu.v4l4j.VideoDevice.doInit(java.lang.String)'
	at au.edu.jcu.v4l4j.VideoDevice.doInit(Native Method)
	at au.edu.jcu.v4l4j.VideoDevice.<init>(VideoDevice.java:271)
	at com.github.sarxos.webcam.ds.v4l4j.V4l4jDevice.getVideoDevice(V4l4jDevice.java:105)
	at com.github.sarxos.webcam.ds.v4l4j.V4l4jDevice.<init>(V4l4jDevice.java:89)
	at com.github.sarxos.webcam.ds.v4l4j.V4l4jDriver.getDevices(V4l4jDriver.java:49)
	at com.github.sarxos.webcam.WebcamDiscoveryService$WebcamsDiscovery.call(WebcamDiscoveryService.java:36)
	at com.github.sarxos.webcam.WebcamDiscoveryService$WebcamsDiscovery.call(WebcamDiscoveryService.java:26)
	at java.base/java.util.concurrent.FutureTask.run(FutureTask.java:317)
	at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1144)
	at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:642)
	at java.base/java.lang.Thread.run(Thread.java:1570)
```

Any help would be appreciated.

alexmao86, 2024-09-02:
this lib is fully tested JDK8 and x86 Linux, windows and armv7. so for later aar64 and etc, this lib is not tested.
on other side, the vl4j bridj .... native code are out of date because of JDK new features.
so use OPENCV driver which uses javacv is strongly recommended. 

thanks

SWSpratlin, 2024-09-02:
I just saw on the documentation that V4L4J was the preferred driver for Raspberry Pi, so I wanted to go through that channel. I can give a shot to OpenCV, but will that cause issues with the Pi hardware?

SWSpratlin, 2024-09-09:
Returning to this issue, as its running into trouble with the 4+ versions of openCV/

SWSpratlin, 2024-09-09:
I think there's an issue with how the videoio.VideoCapture class is being called in the JavaCvDriver class, this is only present in the snapshot versions on maven, will try to build on my own from source

SWSpratlin, 2024-09-09:
Closing this issue as resolved. Will open another potentially. 

Issue comes from the JavaCVDriver class. Needs updating in the import statements to reflect changes in javacpp, javacv, and opencv from bytedeco. 

I don't know how to submit a push/pull request so I'll detail my solution. 

VideoCapture and VideoInput classes need an update to their import statments in the JavaCVDriver class

```
import org.bytedeco.opencv.opencv_videoio.VideoCapture;
import org.bytedeco.videoinput.videoInput;
```
works for me.

closed 2024-09-09 by SWSpratlin

SWSpratlin, 2024-09-14:
I was in fact wrong, that did not work for me. 

The real solution for me was quite different. Context: Running Debian Bookworm on a Raspberry Pi 5. Java 17, Maven installed, Pi Cam Module 3. 

The solution that worked for me was the following: 

- Step 1: Install GStreamer on the Pi. Use [this link](https://qengineering.eu/install-gstreamer-1.18-on-raspberry-pi-4.html) for instructions on how to do this. I used GStreamer 1.22 as that was the most updated version. 
- Step 2: Make sure the `libcamerasrc` is installed as part of GStreamer. Check this with `gst-inspect-1.0 libcamerasrc` if you get your camera's properties, move on. 
- Step 3: Clone the Webcam-capture repo to whatever computer you're developing on
- Step 4: Open the GST1Device.java in a code editor. 
- Step 5: Manually set the Resolution (this could be optional but I have not tested it), Format (RGBx) and Source. 
- - Step 5.5: In the getSource() method, change `final Element source` to the following: `final Element source = ElementFactory.make("libcamerasrc", "libcamerasrc0")`. Next change `source.set()` to the following: `source.set("name", name);`
- Step 6 (Optional): in `open()` add `.append("framerate={whatever you want your framerate to be}/1");` to the end of the other `.append()` statement. Right before `.toString();` save the file.
- Step 7: run `mvn clean package` on the the `driver-gst1` directory. 
- Step 8: add the resulting `.jar` file to your project, however you add local jars to your project. 
- Step 9: set the Driver to GST1Driver as shown in the examples. 
- Step 10: set the camera to `webcam.getDefault();` to use `libcamera`

This SHOULD work, but will likely need troubleshooting on other Pi systems with different packages installed than I currently have. This setup is currently stable for 12+ hour runs. 

This was a long, involved process. Since Pi switched away from the framework that they used to use for their cameras (`/dev/video0, etc.`), you need to adjust your driver files manually to fix these problems.